nordicsemi/Android-BLE-Library #554: I stoped receiving notifications even when i have still not get all data

KevinMartinezC, 2024-04-04:
I am currently using the Android-BLE-library for Kotlin, trying to get the data for switches and APs. When I try to connect a switch, the notifications keep listening, and I get the firmware, serial number, and key. However, when I try it with the AP, I only get the firmware and serial number because it seems the notification stops before getting the key.

`This are my logs from switch `

Reading characteristic 00002a25-0000-1000-8000-00805f9b34fb
gatt.readCharacteristic(00002a25-0000-1000-8000-00805f9b34fb)
onConnectionUpdated() - Device=FC:B9:23:80:01:59 interval=36 latency=0 timeout=500 status=0
Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
Read Response received from 00002a25-0000-1000-8000-00805f9b34fb, value: (0x) 62-63-62-39-32-33-38-30-30-31-35-38
Reading characteristic 00002a26-0000-1000-8000-00805f9b34fb
gatt.readCharacteristic(00002a26-0000-1000-8000-00805f9b34fb)
get serial number bcb923800158
Read Response received from 00002a26-0000-1000-8000-00805f9b34fb, value: (0x) 32-2E-31-64
Writing characteristic ba04c4b2-892b-43be-b69c-5d13f2195392 (WRITE REQUEST)
gatt.writeCharacteristic(ba04c4b2-892b-43be-b69c-5d13f2195392, value=0x0010, WRITE REQUEST)
get firmware 2.1d
Data written to ba04c4b2-892b-43be-b69c-5d13f2195392
Notification received from 0734594a-a8e7-4b1a-a6b1-cd5243059a57, value: (0x) 01-01-74-41-41-41-41-42-33-4E-7A-61-43-31-79-63-32-45-41-41-41-41-44-41-51-41-42-41-41-41-42-41-51-43-35-6F-6C-7A-79-30-62-46-78-35-6E-61-63-77-34-50-6C-66-50-44-78-36-75-76-6D-47-6F-2F-35-30-77-42-6A-66-2B-49-58-35-44-5A-30-76-68-53-69-6A-34-4C-7A-46-47-38-35-2F-49-65-67-75-68-50-66-34-38-4B-47-32-6B-4F-7A-44-39-57-5A-34-6B-31-38-74-72-41-61-43-75-54-76-44-57-4D-41-61-50-31-77-6C-63-38-78-74-63-67-34-32-6A-54-6E-56-78-76-51-76-64-70-55-47-67-6B-33-51-31-75-2B-4C-56-46-45-35-6E-36-35-38-38-39-32-6F-4F-37-67-42-30-75-6E-69-4B-61-55-58-30-39-47-44-54-53-59-73-78-6E-50-36-4F-6B-69-2B-42-52-73-6F-65-73-68-61-58-30-68-44-41-63-54-2F-58-44-4A-78-68-67-33-41-51-63-69-42-6D-50-6E-55-53-49-64-69-46-7A-57-4B-75-48-6D-64
LED turned OFF
Button released
Credits sent successfully
Value 240
RXData 240
Notification received from 0734594a-a8e7-4b1a-a6b1-cd5243059a57, value: (0x) 45-38-41-38-46-44-59-37-73-4B-6D-6C-67-70-35-37-48-51-4E-6E-73-55-63-50-41-63-38-51-36-67-37-56-65-41-4B-62-48-6F-79-6A-6B-39-4E-4A-6D-4E-4C-4B-55-44-42-72-36-75-35-69-37-4D-6C-78-41-45-35-66-53-57-67-48-2F-56-4E-34-51-6C-78-42-5A-68-57-6F-34-33-7A-58-6A-62-44-41-55-55-42-47-74-4D-53-6D-76-46-54-73-36-75-30-63-70-73-67-59-2F-50-47-38-38-62-4F-64-52-47-61-42-54-38-59-44-75-2F-50-44-31-74-2B-6B-64-77-44-03-00-01-00-03-00-01-04
Value 143
RXData 383
gatt.setCharacteristicNotification(0734594a-a8e7-4b1a-a6b1-cd5243059a57, false)
setCharacteristicNotification() - uuid: 0734594a-a8e7-4b1a-a6b1-cd5243059a57 enable: false
Disabling notifications and indications for 0734594a-a8e7-4b1a-a6b1-cd5243059a57
gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x00-00)
get pupkey AAAAB3NzaC1yc2EAAAADAQABAAABAQC5olzy0bFx5nacw4PlfPDx6uvmGo/50wBjf+IX5DZ0vhSij4LzFG85/IeguhPf48KG2kOzD9WZ4k18trAaCuTvDWMAaP1wlc8xtcg42jTnVxvQvdpUGgk3Q1u+LVFE5n658892oO7gB0uniKaUX09GDTSYsxnP6Oki+BRsoeshaX0hDAcT/XDJxhg3AQciBmPnUSIdiFzWKuHmdE8A8FDY7sKmlgp57HQNnsUcPAc8Q6g7VeAKbHoyjk9NJmNLKUDBr6u5i7MlxAE5fSWgH/VN4QlxBZhWo43zXjbDAUUBGtMSmvFTs6u0cpsgY/PG88bOdRGaBT8YDu/PD1t+kdwD
AAAAB3NzaC1yc2EAAAADAQABAAABAQC5olzy0bFx5nacw4PlfPDx6uvmGo/50wBjf+IX5DZ0vhSij4LzFG85/IeguhPf48KG2kOzD9WZ4k18trAaCuTvDWMAaP1wlc8xtcg42jTnVxvQvdpUGgk3Q1u+LVFE5n658892oO7gB0uniKaUX09GDTSYsxnP6Oki+BRsoeshaX0hDAcT/XDJxhg3AQciBmPnUSIdiFzWKuHmdE8A8FDY7sKmlgp57HQNnsUcPAc8Q6g7VeAKbHoyjk9NJmNLKUDBr6u5i7MlxAE5fSWgH/VN4QlxBZhWo43zXjbDAUUBGtMSmvFTs6u0cpsgY/PG88bOdRGaBT8YDu/PD1t+kdwD
Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
onConnectionUpdated() - Device=FC:B9:23:80:01:59 interval=396 latency=5 timeout=620 status=0
Connection parameters updated (interval: 495.0ms, latency: 5, timeout: 6200ms)

`This are for the AP`

Reading characteristic 00002a25-0000-1000-8000-00805f9b34fb
gatt.readCharacteristic(00002a25-0000-1000-8000-00805f9b34fb)
Read Response received from 00002a25-0000-1000-8000-00805f9b34fb, value: (0x) 62-63-62-39-32-33-30-30-63-34-65-38
Reading characteristic 00002a26-0000-1000-8000-00805f9b34fb
gatt.readCharacteristic(00002a26-0000-1000-8000-00805f9b34fb)
get serial number bcb92300c4e8
Read Response received from 00002a26-0000-1000-8000-00805f9b34fb, value: (0x) 32-2E-30-6A
Writing characteristic ba04c4b2-892b-43be-b69c-5d13f2195392 (WRITE REQUEST)
gatt.writeCharacteristic(ba04c4b2-892b-43be-b69c-5d13f2195392, value=0x0010, WRITE REQUEST)
get firmware 2.0j
Data written to ba04c4b2-892b-43be-b69c-5d13f2195392
Notification received from 0734594a-a8e7-4b1a-a6b1-cd5243059a57, value: (0x) 01-01-74-41-41-41-41-42-33-4E-7A-61-43-31-79-63-32-45-41-41
LED turned OFF
Button released
Credits sent successfully
Value 20
RXData 20
onConnectionUpdated() - Device=B9:BC:00:23:EB:C4 interval=36 latency=0 timeout=500 status=16
onConnectionUpdated received status: 16, interval: 36, latency: 0, timeout: 500
Connection parameters update failed with status 16 (interval: 45.0ms, latency: 0, timeout: 5000ms)
onPhyUpdate() - status=0 address=B9:BC:00:23:EB:C4 txPhy=2 rxPhy=2
onConnectionUpdated() - Device=B9:BC:00:23:EB:C4 interval=36 latency=0 timeout=500 status=0
PHY updated (TX: LE 2M, RX: LE 2M)
Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)

philips77, 2024-04-04:
Hello,
Here are the results of my investigation.

## Main issue

I think the main issue is with the MTU. Seems like your switch is requesting high MTU on it's own from the peripheral side.
Have a look at this log entry:
```
Notification received from 0734594a-a8e7-4b1a-a6b1-cd5243059a57, value: (0x) 01-01-74-41-41-41-41-42-33-4E-7A-61-43-31-79-63-32-45-41-41-41-41-44-41-51-41-42-41-41-41-42-41-51-43-35-6F-6C-7A-79-30-62-46-78-35-6E-61-63-77-34-50-6C-66-50-44-78-36-75-76-6D-47-6F-2F-35-30-77-42-6A-66-2B-49-58-35-44-5A-30-76-68-53-69-6A-34-4C-7A-46-47-38-35-2F-49-65-67-75-68-50-66-34-38-4B-47-32-6B-4F-7A-44-39-57-5A-34-6B-31-38-74-72-41-61-43-75-54-76-44-57-4D-41-61-50-31-77-6C-63-38-78-74-63-67-34-32-6A-54-6E-56-78-76-51-76-64-70-55-47-67-6B-33-51-31-75-2B-4C-56-46-45-35-6E-36-35-38-38-39-32-6F-4F-37-67-42-30-75-6E-69-4B-61-55-58-30-39-47-44-54-53-59-73-78-6E-50-36-4F-6B-69-2B-42-52-73-6F-65-73-68-61-58-30-68-44-41-63-54-2F-58-44-4A-78-68-67-33-41-51-63-69-42-6D-50-6E-55-53-49-64-69-46-7A-57-4B-75-48-6D-64
```
On the other hand, the same notification that you get from the AP is only 20 bytes long. 20 bytes is the packet length with the default MTU=23 (max length is 20 bytes). Because of that, the firmware may experience some kind of issue (you try to send a lot more) and it stops.

### Solution

Either make sure you set the same MTU value on both switch and AP or request it from Android.
Add the following:
```kotlin
    override fun initialize() {
         requestMtu(517).enqueue()
         // [...]
```

## Other issues
1. The requests enqueued in the `initialize()` method are first added to the initialization queue before they are started being executed. However, in your code you have the following:
   ```kotlin
        enableNotifications(readChar).done {
            sendCredits()
        }.enqueue()
       // other requests...
   ```
   The `sendCredits()` also enqueues a request, but it will be performed AFTER the initialization is complete. It's not a part 
of the initialization, as it's enqueued outside of `initiazlize()`. It does work, ehy not, but placing it inside that method and in the middle may be confusing. I would recommend just moving it after the last `readCharacteristic` or, if you want to make sure it only happens when notifications are enabled, at least adding some comment.
2. Just a reminder to rename the callbacks. You're handling serial number with LED callback and firmware revision with button callback: 
   ```kotlin
       readCharacteristic(serialNumberCharacteristic).with(ledCallback).enqueue()
   ```

KevinMartinezC, 2024-04-04:
```
 override fun isRequiredServiceSupported(gatt: BluetoothGatt): Boolean {
        val switchService = gatt.getService(BlinkySpec.SWITCH_SERVICE)
        firmwareRevisionCharacteristic = switchService?.getCharacteristic(BlinkySpec.FIRMWARE_CHARACTERISTIC)
        serialNumberCharacteristic = switchService?.getCharacteristic(BlinkySpec.SERIAL_CHARACTERISTIC)

        val sppService = gatt.getService(SPP_UUID)
        readChar = sppService?.getCharacteristic(READ_UUID)
        txCtlChar = sppService?.getCharacteristic(TX_CTL_UUID)
        writeChar = sppService?.getCharacteristic(WRITE_UUID)

        return switchService != null && firmwareRevisionCharacteristic != null && serialNumberCharacteristic != null &&
                sppService != null && readChar != null && txCtlChar != null && writeChar != null
    }
```

KevinMartinezC, 2024-04-04:
```
   @OptIn(ExperimentalUnsignedTypes::class)
    private fun sendCredits() {
        rxCredits = 0x1000
        val data =
            ubyteArrayOf((rxCredits and 0xff).toUByte(), (rxCredits ushr 8).toUByte()).toByteArray()

        writeCharacteristic(txCtlChar, data, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)
            .with { device, data ->
                Log.i("BlinkyManagerImpl", "Credits sent successfully")
                rxCredits = 0
            }
            .fail { device, status ->
                Log.e("BlinkyManagerImpl", "Failed to send credits, status: $status")
            }
            .enqueue()
    }
```

KevinMartinezC, 2024-04-04:
```
   @OptIn(ExperimentalCoroutinesApi::class)
    override fun initialize() {
        requestMtu(508).enqueue()
        // Enable notifications for the button characteristic.

        setNotificationCallback(txCtlChar)
        enableNotifications(txCtlChar).enqueue()

        readCharacteristic(readChar).enqueue()
        // Read the initial value of the write characteristic.
        readCharacteristic(writeChar).enqueue()

        enableNotifications(firmwareRevisionCharacteristic).enqueue()

        // Read the initial value of the button characteristic.
        readCharacteristic(firmwareRevisionCharacteristic).with(buttonCallback).enqueue()

        // Read the initial value of the LED characteristic.
        readCharacteristic(serialNumberCharacteristic).with(ledCallback).enqueue()

        readCharacteristic(serialNumberCharacteristic).with { device, data ->
            val serialNumber = data.getStringValue(0)
            serialNumberValue = serialNumber
            Log.wtf("kevin", "get serial number $serialNumber")
            // Handle serial number value
        }.enqueue()

        readCharacteristic(firmwareRevisionCharacteristic).with { device, data ->
            val firmwareRevision = data.getStringValue(0)
            firmware = firmwareRevision
            Log.wtf("kevin", "get firmware $firmwareRevision")
            // Handle firmware revision value
        }.enqueue()

        setNotificationCallback(readChar).with { device, data ->
            val value = data.value
            Log.wtf("kevin","get data value $value")
            if (value != null) {
                collectDataKey(value)
            }
        }

        enableNotifications(readChar)
            .fail { device, status ->
                Log.wtf("kevin","conextion fail $device and status $status")
            }
            .done {
                sendCredits()
            }.enqueue()
    }
```

KevinMartinezC, 2024-04-04:
```
    private fun collectDataKey(value: ByteArray?) {
        Log.wtf("kevin","get into collect data get value $value")
        value?.size?.let {
            rxCredits += it
        }
        if (value != null) {
            rxData += value
            Timber.tag("ALTA-CHANGED-KEY").d("Value ${value.size}")
        }
        Timber.tag("ALTA-CHANGED-KEY").d("RXData ${rxData.size}")
        if (rxData.size < THREE_INT) return

        val total: Int =
            rxData.copyOfRange(ONE_INT, THREE_INT).fold(ZER0_INT) { a, b -> (a shl 8) + b.toInt() }

        if (rxData.size < total + THREE_INT) return
        val pktType = rxData[ZER0_INT]
        val payload = rxData.copyOfRange(THREE_INT, THREE_INT + total)
        rxData = rxData.copyOfRange(THREE_INT + total, rxData.size)

        when (pktType.toUByte()) {
            1.toUByte()  -> {
                pubKey = payload
                val pubKeyStr = pubKey.decodeToString()

                pupkey  =  pubKeyStr
                disableNotifications(readChar).enqueue()
                Log.wtf("kevin", "get pupkey $pubKeyStr")
//                if (gatt != null) {
//                    keyPub(pubKeyStr, gatt)
//                }
                Timber.tag("ALTA-CHANGED-KEY").d(pubKeyStr)
            }

            3.toUByte() -> {
                val state = payload[ZER0_INT].toInt()
                val inet = if (state and (ONE_INT shl ZER0_INT) != ZER0_INT) INET else EMPTY_STRING
                val mesh = if (state and (ONE_INT shl ONE_INT) != ZER0_INT) MESH else EMPTY_STRING
                val cloud = if (state and (ONE_INT shl TWO_INT) != ZER0_INT) CLOUD else EMPTY_STRING
                val setup =
                    if (state and (ONE_INT shl THREE_INT) != ZER0_INT) SET_UP else EMPTY_STRING
                if (state and (ONE_INT shl THREE_INT) != ZER0_INT) {
//                    if (gatt != null) {
//                        //onDeviceConfigured(gatt)
//                    }
                }
                Timber.tag("ALTA-STATUS").d("state: $inet $mesh $cloud $setup")
            }

            else -> {}
        }
    }
```

KevinMartinezC, 2024-04-04:
The part to get the key stop working on AP while it works on the switch

KevinMartinezC, 2024-04-05:
@philips77 could you provide me an example how it would be the correct way to enable the notifications and read and write properties correctly please

KevinMartinezC, 2024-04-08:
Thank you! I will update it with the comments you added and test it

KevinMartinezC, 2024-04-11:
@philips77 I have a question in the case of the readchart the  `writeCharacteristic `should be done before the `readCharacteristic` or the `readCharacteristic` first? I am new in the BLE word so i would appreciate your help

KevinMartinezC, 2024-04-11:
@philips77 I get this error message when i try to connect an AP using the nRf CONNECT APP 
[image]

KevinMartinezC, 2024-04-11:
Do i need to do a different implementation for android 12?

philips77, 2024-04-12:
No, it should work the same way for any Android 5+ version.

Regarding that error, seems like Android tries to do service discovery or connection param update using Group Type which your device may not support. Perhaps that may be ignored..?

KevinMartinezC, 2024-04-12:
Is it a way to disable it?

KevinMartinezC, 2024-04-12:
When i try to connect with android 12 i get this logs

> unregisterApp() - mClientIf=8
> Connecting...
> gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
> connect() - device: B9:BC:00:23:1F:14, auto: false
> registerApp()
> registerApp() - UUID=cfba8dbc-6cd7-476b-976b-c1d8c9f0801a
> onClientRegistered() - status=0 clientIf=8
> onClientConnectionState() - status=0 clientIf=8 device=B9:BC:00:23:1F:14
> [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
> Connected to B9:BC:00:23:1F:14
> wait(300)
> Discovering services...
> gatt.discoverServices()
> discoverServices() - device: B9:BC:00:23:1F:14
> Request timed out
> Disconnecting...
> gatt.disconnect()
> cancelOpen() - device: B9:BC:00:23:1F:14
> onClientConnectionState() - status=0 clientIf=8 device=B9:BC:00:23:1F:14
> [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
> Disconnected
> gatt.close()
> close()
> unregisterApp() - mClientIf=8

KevinMartinezC, 2024-04-12:
I solved it! I increased the connection timeout for that Android version. I really appreciate your comments and help

philips77, 2024-04-15:
Great to hear that!

closed 2024-04-15 by philips77

KevinMartinezC, 2024-04-18:
@philips77 I have a question for example I have to devices and I want to connect each one when I click but they are in the same ViewModel so it will not navigate to another screen. So when i connect the first one it works perfect but when i will try to connect the second one after finishing adding the first devices. I am not able to connect even if after setting up the first one i disconnect it. Should I stop the scan and start doing another scan again?

KevinMartinezC, 2024-04-19:
this is my current implementation, should i add another method to be able to disconnect and connect a new device ?     

```
override fun release() {
        // Cancel all coroutines.
        scope.cancel()

        val wasConnected = isReady
        // If the device wasn't connected, it means that ConnectRequest was still pending.
        // Cancelling queue will initiate disconnecting automatically.
        // If the device was connected, we have to disconnect manually.
        if (wasConnected) {
            disconnect().enqueue()
        }
    }

```

KevinMartinezC, 2024-04-19:
for the seconde device when i get into here detect it as connect it and it does complete the process
  ```
    setNotificationCallback(readChar).with { device, data ->
            val value = data.value
            if (value != null) {
                collectDataKey(value, address = device.address)
            }
        }
```

    private fun collectDataKey(value: ByteArray?, address: String) {
        value?.size?.let {
            rxCredits += it
        }
        if (value != null) {
            rxData += value
            Timber.tag("ALTA-CHANGED-KEY").d("Value ${value.size}")
        }
        Timber.tag("ALTA-CHANGED-KEY").d("RXData ${rxData.size}")
        if (rxData.size < THREE_INT) return

        val total: Int =
            rxData.copyOfRange(ONE_INT, THREE_INT).fold(ZER0_INT) { a, b -> (a shl 8) + b.toInt() }

        if (rxData.size < total + THREE_INT) return
        val pktType = rxData[ZER0_INT]
        val payload = rxData.copyOfRange(THREE_INT, THREE_INT + total)
        rxData = rxData.copyOfRange(THREE_INT + total, rxData.size)

        when (pktType.toUByte()) {
            ALTA_PUB_KEY -> {
                pubKey = payload
                val pubKeyStr = pubKey.decodeToString()

                pupkey = pubKeyStr
                Timber.tag(ALTA_BLE_TAG).wtf("get pup key %s", pubKeyStr)
                if (pubKey.isNotEmpty()) {
                    pubKeyValue.value = pubKeyStr
                }
                Timber.tag("ALTA-CHANGED-KEY").d(pubKeyStr)
            }

            ALTA_STATE -> {
                val state = payload[ZER0_INT].toInt()
                val inet = if (state and (ONE_INT shl ZER0_INT) != ZER0_INT) INET else EMPTY_STRING
                val mesh = if (state and (ONE_INT shl ONE_INT) != ZER0_INT) MESH else EMPTY_STRING
                val cloud = if (state and (ONE_INT shl TWO_INT) != ZER0_INT) CLOUD else EMPTY_STRING
                val setup =
                    if (state and (ONE_INT shl THREE_INT) != ZER0_INT) SET_UP else EMPTY_STRING
                if (state and (ONE_INT shl THREE_INT) != ZER0_INT) {
                    Log.wtf("kevin","completed")
                    isConnected.value = true
                }
                Timber.tag("ALTA-STATUS").d("state: $inet $mesh $cloud $setup")
            }

            else -> {}
        }
    }

but i am resetting the values before connect into another device 
```
 override fun resetValues (){
        disableNotifications(readChar).enqueue()
        rxCredits = 0x1000
        pupkey = null
        txData = ByteArray(0)
        pubKey = ByteArray(0)
        isConnected.value = false
        firmwareVersion.value = null
        serialNumber.value = null
        pubKeyValue.value = null
        onServicesInvalidated()
        release()
    }
```

```
`  override fun onServicesInvalidated() {
        serialNumberCharacteristic = null
        firmwareRevisionCharacteristic = null
        txCtlChar = null
        readChar = null
        writeChar = null
    }`
```

KevinMartinezC, 2024-04-19:
@philips77 do i need to use another method to clear the notifications?

philips77, 2024-04-19:
Instead of reseting values, like you do, call:
```kotlin
disconnect().enqueue()
```
This will disconnect the device and will call `onServicesInvalidated()` automatically when done. There you may reset your values. I think you're now only disconnecting when the `ViewModel` gets closed, so when you leave the screen, not when you want to connect to another device.

Also, no need to disable notifications, as you're disconnecting after all.

KevinMartinezC, 2024-04-19:
i do it like this after the connection is complete it i call the disconnect method 

>     private val showBottomSheetBLE = combine(
        altaBLE.pubKeyValue,
        altaBLE.firmwareVersion,
        altaBLE.serialNumber,
        altaBLE.isConnected,
        hideBottomSheet,
    ) { key,firm,serial,isConnected,show ->
        if (!key.isNullOrEmpty()  && !firm.isNullOrEmpty() && !serial.isNullOrEmpty() && show) {
            loading.value = true
            deviceConnectionStatus.value = DeviceConnectionStatus.CONNECTING
            selectedBleDevice.value?.model?.let {
                val response = addDeviceUseCase.invoke(
                    siteId = siteId.orEmpty(),
                    serialNumber = serial,
                    pubKey = key,
                    model = it,
                    firmwareVersion = firm
                )
                response.onSuccess { res ->
                    loading.value = true
                    if (!res.setup.isNullOrEmpty()) {
                        altaBLE.confirmSetup(res.setup)
                        //altaLabsBluetoothManager.confirmSetUp(device, res.setup)
                    } else {
                        errorGettingSetUp()
                    }
                    altaBLE.pubKeyValue.value = EMPTY_STRING
                    // altaLabsBluetoothManager.pubKey.value = EMPTY_STRING
                }

                response.onError { _, message ->
                    deviceConnectionStatus.value = DeviceConnectionStatus.ERROR
                    errorGettingSetUp()
                    viewModelScope.launch {
                        delay(ONE_SECOND)
                    }
                    hideBottomSheet.value = false
                    loading.value = false
                    Timber.tag("AddDevice").e(message)
                }

                response.onException { e ->
                    deviceConnectionStatus.value = DeviceConnectionStatus.ERROR
                    Timber.tag("AddDevice").e(e)
                }
            }
        }
        if (isConnected) {
            Log.wtf("Kevin","get into is connected")
            deviceConnectionStatus.value = DeviceConnectionStatus.CONNECTED
            viewModelScope.launch {
                delay(ONE_SECOND_AND_HALF)
                hideBottomSheet.value = false
                loading.value = false
                deviceConnectionStatus.value = DeviceConnectionStatus.DEFAULT
            }
            onRefreshRequest()
            disconnectDevice()
        }
        hid

```
   override fun disconnectDevice() {
        rxCredits = 0x1000
        pupkey = null
        txData = ByteArray(0)
        pubKey = ByteArray(0)
        isConnected.value = false
        firmwareVersion.value = null
        serialNumber.value = null
        pubKeyValue.value = null

        // Cancel all coroutines.
        scope.cancel()
        disconnect().enqueue()
    }
```
it just works connect other device after one stoping the scanner and scan again in other way it get the firmware and serial from the another device but it detect as connect it in the notification as it is taking the data from the previous one in that part

KevinMartinezC, 2024-04-22:
@philips77 I solved it. Thanks. I have another question using that library is able to connect to multiple device at the same time when it is discover by the scanner one by one ?

philips77, 2024-04-22:
Yes, it is possible. You need to have one `BleManager` instance per device. I think modern phones support up to 6 simultaneous connections.

KevinMartinezC, 2024-04-22:
Thank you. I will try it!