nordicsemi/Android-BLE-Library #560: Cannot the BLE device

minhkimlong, 2024-05-01:
I've tried debugging, but I see the function isRequiredServiceSupported do not run when I run with S21 Samsung Galaxy 5G. Any instructions when I work with that device

philips77, 2024-05-08:
Please find the documentation here: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/main/USAGE.md
This method should be called just after the device gets connected.

If you still have issues, please provide logs that you're getting in `log(...)` method.

minhkimlong, 2024-05-08:
Hi @philips77 this is my log
[image]

minhkimlong, 2024-05-08:
I followed all your instructions and tested a lot of Android devices. But only the S21 5G Samsung Galaxy I cannot connect to the Philips diffuser device @philips77

minhkimlong, 2024-05-08:
I have more content for you when I reset all my data and connect to the Philips diffuser device. It took a long time to connect or cannot connect. But if connected successfully in the first time, it will work without any problems. Can I open this issue thank you @philips77

philips77, 2024-05-08:
Hello,
It's hard to say what's the root cause of this issue. From the logs I see the error 0x08 (GATT CONN TIMEOUT), which usually happens when the target device stops responding for _supervision timeout_ milliseconds. That would make sense, as you're getting it 5 seconds (equal to supervision timeout: 5000ms) after the device got connected. 

For me this looks like a phone issue. Either the phone, or the peripheral sends a request and doesn't get a response back and kills the connection after 5 sec. Sometimes it happens when the target does not support a feature, but the central keeps asking for it, or ignores "not supported" and awaits something else.
If you have access to the firmware, try disabling requesting PHY LE 2M or disable support for it and try again. For sure the issue is on the Samsung phone and the best you can do is to find a workaround or stop supporting that model.

minhkimlong, 2024-05-09:
Thank you @philips77 
I will try and get some update info for you