nordicsemi/Android-BLE-Library #561: A couple of questions, possibly bugs

izivkov, 2024-05-07:
I experience a couple of problems, which I think are bugs, at least with version 1.0.14:

In the callback ```override fun onDeviceReady(device: BluetoothDevice)```, ```device.name``` is **sometimes** null. Why?

The callback ```override fun isRequiredServiceSupported(gatt: BluetoothGatt)``` is **sometimes** not called before ```onDeviceReady()```, so my read and write characteristics cannot be initialized properly.

I don't want to describe these as bugs, because there is not enough documentation to determine if I am not using these callbacks properly.

Thank you.

philips77, 2024-05-07:
In [this](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/main/blinky/ble/src/main/java/no/nordicsemi/android/blinky/ble/BlinkyManager.kt) class you have "Blinky" implementation of the Bluetooth LE layer. It is build using this library (Android BLE Library).

Remote device is exposed to the app via an interface named [`Blinky`](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/main/blinky/spec/src/main/java/no/nordicsemi/android/blinky/spec/Blinky.kt). The implementation is private and can be replaced with a mock, for example.

The library does not support scanning, just Client or Client+Server connection mode. Instead, I would recommend 3 options:
1. Using Android Compat Scanner library - if you plan to support Android 4.3+
2. Native Android API (e.g. `BluetoothLeScanner`) - if you like callbacks and builders
3. Kotlin BLE Library 
   * The current version will soon[^1] be replaced with a new implementation

Current version of nRF Blinky is using the first approach, you may find it [here](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/main/scanner/src/main/java/no/nordicsemi/android/scanner/repository/ScannerRepository.kt).

I would recommend option 3 and waiting for version 2.0, if you have time, otherwise any other implementation.

[^1]: I am working on client side at the moment and how to publish something in coming weeks

izivkov, 2024-05-07:
> Footnotes

Awesome, thanks for the info! I will be checking on this repository for
changes.

We can now close this issue if you like.

closed 2024-05-07 by philips77