nordicsemi/Android-BLE-Library #564: Internal write characteistic and dead system exception

talfsaverone, 2024-05-28:
Hi, 
We are using the Nordic SDK and getting the following exception in the Crashlytics:

```
Non-fatal Exception: android.os.DeadSystemRuntimeException
android.os.DeadSystemException
android.bluetooth.BluetoothGatt.writeCharacteristic (BluetoothGatt.java:1501)
no.nordicsemi.android.ble.BleManagerHandler.internalWriteCharacteristic (BleManagerHandler.java:50)
no.nordicsemi.android.ble.BleManagerHandler.nextRequest (BleManagerHandler.java:774)
no.nordicsemi.android.ble.BleManagerHandler.enqueue (BleManagerHandler.java:22)
no.nordicsemi.android.ble.Request.enqueue (Request.java:2)
```

Is it something that is caused by the Nordic SDK or is it just a crash of the OS that by chance happened while running the write characteristic command?

philips77, 2024-05-30:
This seems to be an OS crash. May I ask if you can replicate this issue on your side?

mtuteja-whoop, 2026-01-08:
> > My working theory is a race where we still have an in-flight GATT write/queued operation while the platform Bluetooth stack is tearing down and clearing resources, so we end up hitting a bad state during shutdown.
> 
> I think you're correct. What we could do is add a catch for that exception and shut everything down without crashing.

That’s exactly what we’re planning to do on the app level for now, until the Nordic team patches it in the library. Thanks for quick reply. Let me know if you need anything else from our side.