nordicsemi/Android-BLE-Library #570: Oppo not show paring message

LeugimDev, 2024-07-09:
Hello.

First of all, thank you for the library, I have been using it for a long time and it works better and better.

I have a problem with a device model, specifically an Oppo with Android 14, which, when the Android authorization message has to appear for the user to accept, does not show it and I don't know why.

I have tried other devices, Samsung, Xiaomi, Huawei, LG, Acer... and they all work but not this one.

Specifically, as a curiosity, I have tried different versions of the library and with 2.5.1, if the message appears, with the following ones and not with the last one, and I am interested in using the latest version because I have seen that in the others devices works much better for me.

Do you know any solution to this? Thank you

LeugimDev, 2024-07-09:
To be more exact,

The OS is ColorOS 14.0
Name: OPPO Find X5
Model: CPH2307

philips77, 2024-08-28:
Hello,
The library just reports the broadcast event sent from the system. If the native bonding dialog doesn't show up, it means that something is wrong with the phone. I would suggest the basics: did you try to turn it off and on again? Or at least restart Bluetooth?

We don't have any Oppo devices. Could you please share the log from `log(...)` method in your `BleManager` implementation?

LeugimDev, 2024-08-28:
Hi, I tried that and several other options at the native level, finally I created a method right at the time of pairing where I do the following and it works
in the part of .connect(device)

   ?.before { device ->
                 
   if (device.bondState == BOND_NONE) { 
                                feintBluetoothDeviceDiscovery(device) / 
                            } else {
........}
....
}

    @SuppressLint("MissingPermission")
    private fun feintBluetoothDeviceDiscovery(device: BluetoothDevice) {
        val btAdapter = BluetoothAdapter.getDefaultAdapter()
        GlobalScope.launch {
            delay(1000)
            btAdapter.cancelDiscovery()
            delay(100) 
                   device.createBond() 
      }
       }

philips77, 2024-08-29:
Are you are of the following method:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5627275c4dbf1013b6f1b777bed6d705253ab2b1/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L648-L651
It delays service discovery after successful connection.
Do you request pairing from the device, of from Android? 
If you have an iOS app, there's no API to request pairing from there. You have to request it from the device side, or try to read/write an attribute with permission _encrypted_ or _MITM_.

The above method allows to delay service discovery until you'll get the pairing request from the device side.

LeugimDev, 2024-08-29:
Only use the library in Android.

In some tests I did what you told me, and it didn't work either, doing what I put above was the only thing that worked