nordicsemi/Android-BLE-Library #594: WriteRequest Split : cancelCurrent don't stop sending packets

rlebranchu, 2024-10-10:
Hello,

I'm having a problem cancelling a executing split WriteRequest.
The sending of the packets is not stopped.

I think this is because WriteRequest.hasMore() (in BleManagerHandler.onCharacteristicWrite) will always return the next packet.
If we add this following function to the [WriteRequest](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/main/ble/src/main/java/no/nordicsemi/android/ble/WriteRequest.java) class :
```java
/**
 * Cancel split WriteRequest during sending packets (called by BleManagerHandler.cancelCurrent function)
 * This will block hasMore function called by BleManagerHandler.onCharacteristicWrite function
 */
void cancel() {
	complete = true;
}
```
And call it it [BleManagerHandler.cancelCurrent](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/794e94fd9205dd8b8a2780bd6771abfb5b087cdb/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1674) function (after notifyFail of TimeoutableRequest)
```java
if (request instanceof final TimeoutableRequest r) {
	r.notifyFail(device, FailCallback.REASON_CANCELLED);
} else if(request instanceof final WriteRequest r) {
	r.cancel();
}
```
So when WriteRequest.hasMore() is called, the ‘complete’ variable will be set to true and so will not return the next packet (and will not send the next packet).

LightWeek, 2024-10-10:
@rlebranchu 
Hi. 
To test fork locally you can use local Maven Repo. 
Comment signing in the `publish-module.gradle` and use 
`./gradlew publishReleasePublicationToMavenLocal`

In your project add  `mavenLocal()` in settings.gradle.kts 

```
pluginManagement {
    repositories {
        google()
        mavenCentral()
        mavenLocal()
    }
   }
```

After sync your project will use your locally built library.

philips77, 2024-10-10:
Hi,
Yes, that's a known issue. The `WriteRequest` is not cancelable (as apps should always wait for `onCharacteristicWrite` callback), but as `split` was added, and the request can be composed of multiple write operations, it should be cancellable.
As I'm implementing Kotlin version of BLE library ([Kotlin BLE Library, branch: version/2.0](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/tree/version/2.0)), I'm trying not to add more features here. But I'll check how much work would it be needed to fix this issue. The same with `ReadRequest`.

philips77, 2024-10-10:
OK. It took me 18 min. I planned 2 days. My time estimation is really bad.
However, I did not test this new feature, so I'm relaying on you :) As there was very little changes, the previous behavior should not be affects. By extending `TimeoutableRequest`, each class received `timeout(...)` and `cancel()` methods.

philips77, 2024-10-10:
I'll release it as _2.7.0-beta03_.

closed 2024-10-10 by philips77

rlebranchu, 2024-10-10:
Ok thanks! I'll test it when I get back to work tomorrow and confirm if it fixes the problem.

philips77, 2024-10-11:
I'm trying to fix #552 (the other issue from that issue), after that I'll push it. I was hoping today, but I am testing on 6 phones, and it takes time.

One thing. I recommend using a bit different indentation in your code:
```kotlin
val response = waitForNotification(wCommandChar)
   .trigger(
      writeCharacteristic(wOTAChar, otaData, BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE)
         .split(
             WriteProgressCallback { device, data, index ->
                // code..
             }
         )
         .before { device ->
             // code..
         }
         .done { device ->
             // code..
         }
         .fail { device, status ->
             // code..
         }
   )
   .timeout(10000L)
   .suspendForResponse<ReadResponse>()
```
When I read your's I thought you're assigning `.done`, '.fail` and `.before` to the `waitForNotification`, not the triggering call. But perhaps it's just me and you have your style :)

philips77, 2024-10-11:
https://github.com/NordicSemiconductor/Android-BLE-Library/releases/tag/2.9.0-beta03 published