nordicsemi/Android-BLE-Library #602: Which library `Android-BLE-Library` or `Kotlin-BLE-Library` to choose for production project starting today

BhavyaRattan, 2025-01-01:
Hi

Thanks for creating the amazing libraries for BLE on Android, I am working on a production project to be released on playstore in early April & a bit confused between which BLE library `Android-BLE-Library` or `Kotlin-BLE-Library` to use in my project. Most of the github issues on `Kotlin-BLE-Library` point to using `version2.0` but that version is still in early dev & not feature complete. I am tempted to use `Android-BLE-Library` since it has been battle tested earlier on prod projects by the community but also a bit skeptical regarding the upcoming support & maintenance for it. Do you plan to actively maintain & fix upcoming issues for `Android-BLE-Library` while the Kotlin library version 2.0 is ready for production?

philips77, 2025-01-07:
Hello!
Yes, at least for many months we will support _Android BLE Library_.

If you only need a client, I recommend Kotlin BLE Library v2, as you said. Seems like all the basics are working. The API isn't stable yet, needs to be more [DRY](https://pl.wikipedia.org/wiki/DRY), but we're getting there.

Anyway, if you use the `BleManager` as a middle layer between your app and the device API, transitioning should be easy afterwards. Don't expose library API from the manager.

satsen, 2025-01-30:
My Android project is written in Java, does that mean I cannot use any of your libraries because this one will be discontinued and the other is for kotlin?

philips77, 2025-02-04:
We are still supporting this library, fixing issues, etc. It is released on Maven Central, from which it is not possible to remove anything, so all your builds will work. 

We're not planning to add more features here, but I believe the lib is feature complete already. Feel free to use this one for Java projects and the new one for your new Kotlin projects. I believe the new one will have a nicer API and be way more test-friendly.