nordicsemi/Android-BLE-Library #603: waitForNotification with multiple notifications

ThijnSmulders, 2025-01-13:
As the title suggests, I have a device which sends me anywhere from 1 to 100 notifications when i send a command to it. The end of this data stream is indicated by a special series of bytes. How would i use the `waitForNotification` method (or anything similar) to keep receiving and storing these notifications until i detect the special bytes? I suspect it is a combination of `suspendForValidResponse` and `merge` but i can't quite figure it out.

philips77, 2025-01-13:
There's also `filter` and `filterPacket`, which you may find useful.

ThijnSmulders, 2025-01-14:
Yes that helped, I had misunderstood the functionality of `merge` (I missed the fact that you can tell it to wait for more data with the return boolean). It all works flawlessly now, great library!

closed 2025-01-14 by ThijnSmulders