nordicsemi/Android-BLE-Library #604: readcharacteristic only accepts data once

0d3ng, 2025-01-25:
Hello,
currently I am using library version 2.5.1. Why does readCharacteristic only receive data once to receive data from BLE devices? is there anything that is still not right?

`public class MyBleManager extends BleManager {
    private static final String TAG = MyBleManager.class.getSimpleName();

    private BluetoothGattCharacteristic targetCharacteristic;

    public MyBleManager(@NonNull Context context) {
        super(context);
    }

    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
        return new MyBleManagerGattCallback();
    }

    private class MyBleManagerGattCallback extends BleManagerGattCallback {
        @Override
        protected void initialize() {
            super.initialize();
            Log.d(TAG, "Initializing BLE connection...");
            requestMtu(50)
                    .done((device -> Log.d(TAG, "initialize: " + device.getAddress())))
                    .fail(((device, status) -> Log.d(TAG, "initialize: " + device + " :" + status)))
                    .enqueue();
            if (targetCharacteristic != null) {

                setNotificationCallback(targetCharacteristic)
                        .with((device, data) -> {
                            String value = data.getStringValue(0);
                            Log.d(TAG, "onDataReceived: " + value);
                        });

                enableNotifications(targetCharacteristic)
                        .done(device -> Log.d(TAG, "Notifications enabled on: " + targetCharacteristic.getUuid()))
                        .fail((device, status) -> Log.e(TAG, "Failed to enable notifications, status: " + status))
                        .enqueue();

                readCharacteristic(targetCharacteristic)
                        .with((device, data) -> {
                            byte[] dataValue = data.getValue();
                            Log.d(TAG, "initialize data: " + data.toString());
                        })
                        .fail((device, status) -> Log.e(TAG, "Failed to read characteristic, status: " + status))
                        .enqueue();
            } else {
                Log.w(TAG, "Target characteristic is null, cannot initialize notifications.");
            }
        }

        @Override
        protected boolean isRequiredServiceSupported(@NonNull BluetoothGatt gatt) {
            Log.d(TAG, "Discovering services...");
            List<BluetoothGattService> services = gatt.getServices();

            for (BluetoothGattService service : services) {
                Log.d(TAG, "Found service: " + service.getUuid());
                List<BluetoothGattCharacteristic> characteristics = service.getCharacteristics();
                for (BluetoothGattCharacteristic characteristic : characteristics) {
                    Log.d(TAG, "Found characteristic: " + characteristic.getUuid());

                    if ((characteristic.getProperties() & BluetoothGattCharacteristic.PROPERTY_READ) > 0) {
                        Log.d(TAG, "Characteristic supports READ: " + characteristic.getUuid());
                    }
                    if ((characteristic.getProperties() & BluetoothGattCharacteristic.PROPERTY_NOTIFY) > 0) {
                        Log.d(TAG, "Characteristic supports NOTIFY: " + characteristic.getUuid());
                        targetCharacteristic = characteristic;
                    }
                }
            }

            boolean isSupported = targetCharacteristic != null;
            Log.d(TAG, "Required service supported: " + isSupported);
            return isSupported;
        }

        @Override
        protected void onServicesInvalidated() {
            Log.d(TAG, "Services invalidated. Cleaning up...");
            targetCharacteristic = null;
        }

    }

    int toInt(byte b1, byte b2) {
        return ((b1 & 0xFF) << 8) | (b2 & 0xFF);
    }

    float toFloat(byte b1, byte b2, float scaleFactor) {
        int intValue = toInt(b1, b2);
        return intValue / scaleFactor;
    }
}`

I tried reading data using nRF Connect can receive data every 1 minute. BLE devices are for reading air quality sensors.
Thank you very much

[image]

philips77, 2025-01-29:
Hello,
Reading is a one-time operation.
What you do in nRF Connect, is you _enable notifications_ on this characteristic and then the app is updated each time the device changes its value.
You do this here:
```java
                setNotificationCallback(targetCharacteristic)
                        .with((device, data) -> {
                            String value = data.getStringValue(0);
                            Log.d(TAG, "onDataReceived: " + value);
                        });

                enableNotifications(targetCharacteristic)
                        .done(device -> Log.d(TAG, "Notifications enabled on: " + targetCharacteristic.getUuid()))
                        .fail((device, status) -> Log.e(TAG, "Failed to enable notifications, status: " + status))
                        .enqueue();
```

You should be getting "onDataReceived: <some text>".

> [!Note]
> Also, in nRF Connect, if a characteristic returns a String, you may long-tap it, select Edit icon in the menu and set the format to TEXT to have it parsed.

0d3ng, 2025-01-30:
thank you very much @philips77 for your guide.  The problem is solve

closed 2025-01-30 by philips77