nordicsemi/Android-BLE-Library #605: Using asFlow() fro no.nordicsemi.android.ble.ktx

natalyabl, 2025-01-31:
Hi!
Last year we upgraded our Android app to use android.ble.ktx:2.7.5  to communicate with sensors we develop. We are working on upgrading to 2.9.0 but there are a few extra changes we have to do on our side. 
Recently we've noticed an increase on delays in the communication and discovered that most of the time some packages are going missing. 

We decided to use asFlow to get notifications from the sensor like this: 
        notificationsFlow = setNotificationCallback(readCharacteristic).asFlow()
        enableNotifications(readCharacteristic).enqueue()

which we call in the initialize function. 
In our app, we scan for the sensor, connect to it and then we request some information from the sensor which we collect from notificationsFlow. For some reason, the first packet we are expecting, sometimes is not received. Is there an specific way of doing this? 
I haven't found an example using asFlow(), could somebody maybe send across an example or is it not reliable to be use as it's annotated with @ExperimentalCoroutinesApi?

Thanks!
Nbl

philips77, 2025-02-04:
Hello,
You're using it correctly. Let me check if there are no bugs on our side.

philips77, 2025-02-04:
Could you do an experiment? 
Have a look at this code:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/950451532a94f101344c0426eebac565472aeb8b/ble-ktx/src/main/java/no/nordicsemi/android/ble/ktx/ValueChangedCallbackExt.kt#L24-L34
In line 28 we're calling `trySend(..)`, which may fail, if the buffer is full (last event wasn't consumed). I suspect that this is the reason why you would loose packets. The documentation for `callbackFlow` says:
> A channel with the [default](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.channels/-channel/-factory/-b-u-f-f-e-r-e-d.html) buffer size is used. Use the [buffer](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/buffer.html) operator on the resulting flow to specify a user-defined value and to control what happens when data is produced faster than consumed, i. e. to control the back-pressure behavior.

Could you add `buffer(capacity = ...)` on the resulting flow and check if it helps?

Another option would be to replace `trySend(data)` with `trySendBlocking(data)`, which block the sender when the buffer is full. If you have the library added as a module from source, could you modify the source and test it?

philips77, 2025-02-04:
Could you try the PR made and
1. Check if number of lost packets dropped to 0
2. Paste LogCat logs if it didn't

natalyabl, 2025-02-05:
Hi! 
thanks for your quick reply! 
I've tried the PR and it seems to be working alright but I also noticed a few other problems with our implementation that I had to change. 
I'll try both `trySend` and `trySendBlock` with the new implementation and come back to you. 
Thanks a lot for your help! 
Nbl

PR #607 (2025-03-04): Sending notifications as Flow more reliably
This PR fixes #605. It improves how the data recieved in Notifications are sent to the `Flow` by using `trySendBlocking` instead of `trySend`, which makes only one attempt to send the data. If the consumer is busy, the received data were dropped.
closed 2025-03-04 by philips77