nordicsemi/Android-BLE-Library #615: The error "TOO MANY OPEN CONNECTIONS" is returned in onDeviceFailedToConnect

LuoPeiQin, 2025-03-28:
Reproduction environment:
Phone: Google Pixel 6 (other Google phones should be the same)
Android version: Android 14

Reproduction steps:
Method One:
1. Connect to a Bluetooth device and disconnect after ten seconds;
2. Wait ten seconds and then reconnect to this Bluetooth device, then loop steps 1 and 2;
3. After repeating steps 1 and 2 about twenty times, the "TOO MANY OPEN CONNECTIONS" message will appear, and at this point, no Bluetooth devices can be connected unless Bluetooth or the App is restarted;
Method Two:
1. Simultaneously connect multiple devices and disconnect after ten seconds;
2. Reconnect multiple devices and repeat steps 1 and 2;
3. The issue can be reproduced within about five minutes;

I did not encounter this problem on an OPPO phone with the latest Android 15 system. Could this issue be related to any recent code changes by Google? How should I resolve this issue? 

I am eagerly awaiting replies from everyone and the author. Thank you very much.

LuoPeiQin, 2025-03-28:
After the issue occurred, I checked the Heap Dump of the Profiler and found that all BleManager instances were held by BluetoothGatt, but in fact, the disconnect calls were all BleManager.
[image]

philips77, 2025-04-01:
Hi,
Do you know what type is `BleManagerHandler$4` exactly?
This seems to be a list / map holding instances of `BluetoothGatt` objects.

LuoPeiQin, 2025-04-09:
@Override
	public void postDelayed(@NonNull final Runnable r, final long delayMillis) {
		new Timer().schedule(new TimerTask() {
			@Override
			public void run() {
				r.run();
			}
		}, delayMillis);
	}

After comparative analysis, it was found that BleManagerHandler$4 corresponds to the code mentioned above, but the specific cause has yet to be determined.

Thank you very much for providing such an excellent open-source library.