nordicsemi/Android-BLE-Library #617: Connection - Disconnection question

brakinhoo7, 2025-04-03:
Hi, 

First of all thank you for your amazing library which is very helpful.

I'm working on a device that includes a custom BLE device and work with an Android app I'm developing too, I have some questions regarding the connection processus with a Samsung device. 

First, when I want to connect to a custom BLE device, the Samsung system always show a popup asking if I want to launch a pair request or if I want to cancel it. Here, when I click on "Pair", the system is asking me a passkey to connect (which for the moment is a static passkey, I want to go with random passkey as soon as I will fix issues with this pairing process). But if I click on cancel, the behavior is quite weird:

I'm listening to the connection state through `BleManager.stateAsFlow()`:

```
getStateAsFlow()
      .collect {
             Timber.d("Connection state changed to $it")
             if (it.isReady) {
                  Timber.d("Bond state: $bondingState")
                  if (isBonded) {
                       Timber.d("ok")
                  } else {
                       Timber.d("trying to disconnect")
                       disconnectDevice()
                 }
            }
```
When I click on "cancel", the connection state change to ConnectionState.Ready while the bonding state is either Bonding or NotBonded. That is why I try to disconnect with `BleManager.disconnect()` if the pairing process was canceled.

```
override suspend fun disconnectDevice() = super.disconnect()
        .before { device ->
            Timber.d("Disconnecting from $device")
        }
        .done { device ->
            super.close()
            Timber.d("Disconnected from $device")
        }
        .invalid {
            Timber.d("Disconnection could not be executed")
        }
        .fail { device, status ->
            Timber.d("Failed to disconnect from $device with status $status")
        }
        .suspend()
```

But here, it appears that I never go on the done callback of `BleManager.disconnect()`, and the ConnectionState is never changet to Disconnecting and Disconnected. See my logs below:

```
D  Connecting to device XX:XX:XX:XX:94:14
D  Connection state changed to Connecting
I  BleManager: Connected to F6:2F:D3:98:94:14
D  Connection state changed to Initializing
D  BleManager: [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3); key: -1
I  BleManager: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I  BleManager: Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
W  BleManager: Bonding failed
I  BleManager: Services discovered
D  initialize
D  bondState: 10
D  Bond created
D  Connection state changed to Ready
D  Bond state: Bonding
D  trying to disconnect
D  readNotifyFlow
I  BleManager: MTU changed to: 90
D  Disconnecting from XX:XX:XX:XX:94:14
I  BleManager: Disconnected
W  onServicesInvalidated
W  Can't request high priority
```

Do you have any advice or any help to provide me in order to optimize my connection-disconnection processus? I would be very grateful.

Thank you for your work and your helpness!

philips77, 2025-04-07:
Hi,
This seems to be a bug. I'll have a look at the code, thank you for bringing this up.

I could suggest 2 improvements that you can try, in addition.
1. Instead of requesting Bonding from the device just after connection, try securing your attributes with _protected_ or _protected MITM_ permissions. In that case bonding will be triggered when the connected client tries to access the attribute, i.e. enable notifications or write / read some characteristic. 
2. When bonding fails, for any reason, disconnect from the device side. You should be getting a callback on the fw side with soome failed status, from which you can disconnect. If bonding is required for your device to operate and be secure, it is the device that should ensure that. Otherwise it would be possible for other apps, like nRF Connect, to connect to it, cancel pairing and use attributes without security.

philips77, 2025-04-07:
Did you try calling `enqueue()` instead of `suspend()` in your `disconnectDevice()` method?

My guess is, that you're still suspended in "disconnecting", so the state flow can't emit the new state.

philips77, 2025-04-07:
Also, the `suspend()` method is internally using `done(..)` and `fail(..)` callbacks, so it overwrites the one you have set. For suspended method you can just do try-catch to get a failure and whatever you'd have in `done(..)` just place it after the method. It's blocking, so when it's done it will just resume.

brakinhoo7, 2025-04-07:
Hi,

Thank you for your reply!

> Did you try calling `enqueue()` instead of `suspend()` in your `disconnectDevice()` method?
> 
> My guess is, that you're still suspended in "disconnecting", so the state flow can't emit the new state.

That was a good guess! By calling `enqueue()` callbacks are executed. I should have think of that, it was quite obvious haha! Thank you for bringing this up.

> Hi, This seems to be a bug. I'll have a look at the code, thank you for bringing this up.
>
> I could suggest 2 improvements that you can try, in addition.
>
> 1. Instead of requesting Bonding from the [...]

1. I've done that and it works perfectly fine. In fact we've already secured our attributes with _protected MITM_ permission, but I don't know why we implemented the request Bonding just after the connection also.
2. This is hard for us to implement this because our custom BLE device is developed to work as the slave, so we have to completely change the profiles of both BLE device and Android app. Unfortunately we don't have much time right now to do this.. But I see the point and that is for sure one of the first improvements to develop.

What is the bug your calling here? Is it the fact that we got to `ConnectionState.Ready` even if the pairing request was cancelled?

philips77, 2025-04-08:
> What is the bug your calling here? Is it the fact that we got to ConnectionState.Ready even if the pairing request was cancelled?

No, I was thinkin of the status not changing to discopnnected. But as you were using `suspend()`, this may cause issues, as the call emits a new state but the collector is blocked, so perhaps you don't collect it. 
Now, when it works with `enqueue()` I don't think there are any issues. I checked the code and the callbacks are called.

Anyway, great that you solved your issue. Good luck with the project :)

closed 2025-04-08 by philips77