nordicsemi/Android-BLE-Library #626: Question about auto connect

minwang1, 2025-07-07:
Hello! First off, thanks for this awesome library.

I've run into an issue using the auto-connect feature: When I set autoConnect() to true and successfully connect to a device, if I then turn off Bluetooth through the system while still connected, I receive a REASON_TERMINATE_LOCAL_HOST callback in onDeviceDisconnected(). However, when I re-enable Bluetooth, it doesn't auto onnect again. I haven't called the BluetoothDevice's createBond() method.

Is this a bug, or do I need to implement some code to handle this scenario?

minwang1, 2025-07-08:
I have handle this scenario, thanks.

closed 2025-07-08 by minwang1

philips77, 2025-07-09:
Hi,
Sorry, I missed your question. Autoconnect only works until you turn off the Blluetooth.
For bonded devices it's possible to get list of them and call `connectGatt(autoconnect=true)` and it will connect.
For not bonded devices the phone needs to scan them before connecting. If you just save the MAC address and try to create `BluetoothDevice` manually and connect, the phone may assume public address type and won't connect to a device with random address, as the 49th MAC bit won't match. Scanning updates the internal database on the phone, so the phone will know the address type for the connection.

[This](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getRemoteLeDevice(java.lang.String,%20int)) new method added option to create a `BluetoothDevice` with random address type, so it may work on API 33+. I haven't had the opportunity to test it yet.

minwang1, 2025-07-09:
Yes, thanks for reply, I do some study about bluetooth MAC address in android.
Typically, there are 4 types of mac address
1. Public Device Address --once you scaned, you can save MAC address and use it for reconnect.
2. Static Random Device Address --may need to scan, device MAC may change after reboot.
3. Non-Resolvable Private Address --need to scan every time.
4. Resolvable Private Address --If bond, you can save MAC and use for reconnect, if no, need to scan every time.

minwang1, 2025-07-09:
That helps, thanks for your comment.