nordicsemi/Android-BLE-Library #627: Central device’s GattServer may disconnect the Peripheral’s GattClient after 13 seconds

haogev6, 2025-07-18:
I conducted an experiment and preliminarily identified a potential logic issue. The experimental procedure was as follows:

1. Prepared an Android Central device (OPPO K10 5G, Android API 34) and an Android Peripheral device (Xiaomi 10, Android API 31). Both devices implemented GattClient and GattServer roles.
1. Instructed the Central device to scan for the Peripheral device’s broadcasts and establish a connection.
1. Within the Peripheral device’s GattServer onDeviceConnectedToServer method, created a BleManager object and called useServer and connect methods to establish a reverse connection from the Peripheral’s GattClient to the Central’s GattServer.
1. The Central’s GattServer onDeviceConnectedToServer received the Peripheral’s access event, then created a BleManager object and called useServer and attachClientConnection methods.
1. Surprisingly, approximately 13 seconds later, the Central’s GattServer onDeviceDisconnectedFromServer received a disconnection event from the Peripheral’s GattClient.

I believe the root cause may lie in the BleServerManager.useConnection method. 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/d5b549c71d1c461df142e55487006dbc7180d11f/ble/src/main/java/no/nordicsemi/android/ble/BleServerManager.java#L173-L178
The condition for calling server.connect here appears should be when SDK_INT < S **AND** BLUETOOTH_CONNECT permission is granted, rather than when SDK_INT < S **OR** BLUETOOTH_CONNECT permission is granted.

At least, after replacing **||** with **&&**, the aforementioned issue no longer occurs.

Please verify if my judgment is correct. Thank you.

closed 2025-07-18 by haogev6

philips77, 2025-07-18:
Before Android S there is no `BLUETOOTH_CONNECT` permission.

Anyway, thanks for using the library!
If you have some time, please try our new [Kotlin BLE Library](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/tree/version/2.0). Any feedback is welcome! It doesn't support server side yet, so may not meet your requirements yet.

philips77, 2025-07-18:
We plan to maintain the Java library. I'm not sure if we will add more features, but again - it's feature complete now.
There are problems with it:
* Java,
* Testability, or rather lack of it,
* All BLE implementation basically in a single file.

The `ble-ktx` module added some nice Kotlin features, so it's not terrible :)

The Kotlin one has way better design, we plan to have it as KMP. It has mocking build in with emvironment and peripheral simulation. And there's no Java!

We used to have all apps using this library, now migrating towards the Kotlin one. Eventually it will mature.